lorol/arduino-esp32fs-plugin #4: Binary folder on Window

cotestatnt, 2021-05-29:
Hi @lorol !
Thanks for this plugin, it's really wath is missing with Arduino IDE.

I'm not able to upload data folder using **FatFS**
After downloaded mkftafs binary, i've created a folder in (the same where for example is already instaled and working **mkspiffs** tool)
_**C:\Users\xxxxc\AppData\Local\Arduino15\packages\esp32\tools\mkfatfs\mkfatfs.exe**_

Arduino IDE give me always the error message:
**FatFS Error: mkfatfsnot found!**

What I'm doing wrong?

Thanks.

lorol, 2021-05-29:
Try to put mkfatfs.exe one folder up

cotestatnt, 2021-05-29:
I've tried, no way. 
Both up and down level.

You could add the complete path to error message, so can be easy for the user understand where binary must be located.

jaggil, 2021-06-26:
@cotestatnt Hello, in my case I got it in this folder: C: \ Users \ xxxx \ AppData \ Local \ Arduino15 \ packages \ esp32 \ hardware \ esp32 \ 1.0.6 \ tools \ mkfatfs
Inside I put::
libgcc_s_sjlj-1.dll
libstdc++-6.dll
libwinpthread-1.dll
mkfatfs.exe

Finally I put the following in the Arduino project folder:
E: \ xxxxxx \ Arduino \ tools \ ESP32FS \ tool \esp32fs.jar
I had to capitalize the name \ ESP32FS \

cotestatnt, 2021-06-28:
Thank you so much @jaggil !!!
Finally i've got this plugin working
For me was enough putting the binary **mkfatfs.exe** in the folder suggested

closed 2021-07-31 by lorol