lorol/arduino-esp32fs-plugin #15: ESP32FS - esptool not found

Zyrem, 2023-06-24:
Arduino IDE 1.8.19
ESP32FS 2.0.7
LittleFS_esp32 1.0.6
ESP32C3 Dev board

MacBook Pro 2012 - 10.15.7
MacBook Pro 2020 - 12.6.6

Issue occurred when try to upload file using ESP32FS
error reported esptool not found.

Tried upload using Windows has no issue

3Dsf-info, 2023-07-27:
This problem has existed for a while. Answer here:

https://github.com/me-no-dev/arduino-esp32fs-plugin/issues/15

Have a look at the solution from moononournation.

Zyrem, 2023-07-28:
> This problem has existed for a while. Answer here:
> 
> [me-no-dev#15](https://github.com/me-no-dev/arduino-esp32fs-plugin/issues/15)
> 
> Have a look at the solution from moononournation.

Thank you for the reply.

I have successfully make SPIFFS(from me-no-dev) upload works and carefully check that the issue persist with LittleFS on MacOS

siteswapjuggler, 2024-01-29:
Same here the problem persist with LittleFS on MacOS
"Error: esptool not found!"