lorol/arduino-esp32fs-plugin #16: ESP32 Sketch Data Upload menu item not appearing despite correct installation

Danial-Sy, 2023-10-22:
I installed the file exactly how it was meant to be installed, with the location being sketchbook directory\Documents\Arduino\tools\ESP32FS\tool\esp32fs.jar but nothing is appearing. I've tried reinstalling it and restarting my computer but thus far nothing has gotten it to show up. Any idea why this is happening?

DzenDyn, 2023-12-10:
i have same problem

Grey-Lancaster, 2024-01-06:
same problem

[image]

[image]

Grey-Lancaster, 2024-01-06:
found this, use an old version and it worked

https://github.com/me-no-dev/arduino-esp32fs-plugin/issues/15
[image]

Megumin6626, 2024-09-13:
Got the same problem. For me, it's the newer `mkfatfs.exe` and `mklittlefs.exe` causing the problem. Solved it by using the older version uploaded on Oct 9, 2020. I used the latest release of `esp32fs.jar` from **esp32fs.zip** (release Feb 3, 2021)

And from the older release **[Added a choice to "Erase All Flash"]** (release Oct 9, 2020) Under the assets, download and unzip:

**- mkfatfs.zip**
**- x86_64-w64-mingw32-mklittlefs-295fe9b.zip**

[image]
[image]

aIecxs, 2024-09-29:
did not work for me on Windows 7 Professional SP1 x64 (Build 7601)
extracted [x86_64-w64-mingw32-mklittlefs-295fe9b.zip](https://github.com/lorol/arduino-esp32fs-plugin/releases/download/2.0.3/x86_64-w64-mingw32-mklittlefs-295fe9b.zip) v2.0.3 to
```
%UserProfile%\Documents\Arduino\tools\esp32fs\tool\mklittlefs.exe
%UserProfile%\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933\mklittlefs.exe
```
and [mkfatfs.zip](https://github.com/lorol/arduino-esp32fs-plugin/releases/download/2.0.3/mkfatfs.zip) v2.0.3 to
```
%UserProfile%\Documents\Arduino\tools\esp32fs\tool\mkfatfs.exe
```
@Megumin6626 where to place the binaries?